joelsernamoreno/EvilCrow-Keylogger #5: Nice Project, here is a few things I would like to see

bmatusiak, 2020-08-14:
1. Product Datasheets and PCB pinout images
2. I see you are updating the ESP32 via a Arduino UNO.. but i think it possible to use the the ATMEGA32U4 chip that is already onboard. 
3. Need to List Chip Specs.. like speed, ram, flash. 
4. add in "replay" to send keystrokes back. like rubber ducky.
5. other HID device support, (raw hid sniffing stream)

thanks for letting me voice my opinions, i just ordered one. can't wait to contribute

bmatusiak, 2020-08-14:
honestly i was about to order pretty much a Atmega32U4 BadUSB+ESP32+MicroSD bard, and a USB Host MAX3421.  then found this.

joelsernamoreno, 2020-08-14:
> 1. Product Datasheets and PCB pinout images
> 2. I see you are updating the ESP32 via a Arduino UNO.. but i think it possible to use the the ATMEGA32U4 chip that is already onboard.
> 3. Need to List [...]

Hello!

Thank you very much!

Well, read the whole repository again, the device can be flashed with a script through atmega32u4. You have 3 installation methods: automatic install, automatic install with ESPFlasher and manual install

On the pinout, you have the pins of ESP32-PICO, I have to upload the pins of atmega32u4

And add more features... atmega32u4 has a limited memory and not much space left, it's a bit difficult to add the rubber ducky features if you want to have a Keylogger at the same time, but... you can try :D

bmatusiak, 2020-08-15:
yea i hear ya on the space, its always a factor..  but the microsd has potential for storing the duck scripts. 

ive played around with this, 

https://usa.banggood.com/CJMCU-Badusb-USB-Micro-SD-Virtual-Keyboard-ATMEGA32U4-Support-TF-Memory-p-1203001.html
and it follows the source, 
https://github.com/mharjac/bad_ducky/blob/master/bad_ducky.ino

, and i found this, https://github.com/puckk/CJMCU-3212 that has instructions to flash a ESP8266 from a atmega32u4

bmatusiak, 2020-08-15:
and for raw HID logging..  some devices are  hid-raw/serial .. and this would be super useful for debugging that data to a browser via stream (this is my original idea/use case)..   

if somehow i can get that, it would be a awesome tool for debuging stuff in field without extra hardware (so i wont need a Beagle USB 12 Protocol Analyzer)

joelsernamoreno, 2020-08-15:
Yeah, if you look in my github you can find codes for CJMCU-3212