joelsernamoreno/EvilCrow-Keylogger #7: Problema

rocolobo, 2020-11-30:
Hi Joel,

I bought the Keylogger device. I have a virtual machine with VirtualVox and Ubuntu. I'm a very newbie ... I'm following your instructions, but at this point I get lost:

"Set up a new layout:

0.- Open Keyboard/src/Keyboard.cpp with a text editor"

Where is that path? Should the keylogger device be connected to the PC and the keyboard in this step?

Bests regards

rocolobo, 2020-11-30:
[image]

joelsernamoreno, 2020-11-30:
Hi!

You have to download the Evil Crow Keylogger repository in your Ubuntu VM

Inside this directory you can find the file keyboard.cpp

joelsernamoreno, 2020-11-30:
If you don't get this, I can connect to your computer with teamviewer and fix this

joelsernamoreno, 2020-11-30:
If you need this, write me a private message on twitter (@JoelSernaMoreno)

Thanks!

rocolobo, 2020-11-30:
Thanks Joel¡¡¡How fast¡¡¡

Would it be ok like this?:

[image]

Sorry, but I don't have social networks ... Is there any way to contact you by email?

Bests Regards

rocolobo, 2020-11-30:
[image]

rocolobo, 2020-11-30:
Which one of the three ways of installation do you recommend?

rocolobo, 2020-11-30:
I have followed the steps of the first method (Automatic installation), but it gives the following error:

[image]

I have edited the file and I have changed pio to pio3, which is the command I had used to do this:

sudo pip3 install pyserial

but the same error comes out...

I think everything is well installed:

[image]

rocolobo, 2020-11-30:
In order to use the pio command, I have done the following:

sudo apt install platformio

But when executing the file ./flash.sh it shows me the following error:

[image]

joelsernamoreno, 2020-11-30:
Hello!

Do you have telegram? You can contact me by telegram (@JoelSerna)

joelsernamoreno, 2020-11-30:
Btw... do you have the Evil Crow Keylogger repository in the same directory as keylogger-pio?

rocolobo, 2020-11-30:
Hi,

No, I do not have. I'm sorry... I only have WhatsApp and email

I have the two folders inside my user (lardin). It is so, right?

rocolobo, 2020-11-30:
Do I have to install pip for Python3 or for Pyton2?

joelsernamoreno, 2020-11-30:
I think this is not necessary

Well, send an email here:

joelsernamoreno@gmail.com

rocolobo, 2020-11-30:
I was missing adding pio to environment variables. Now it has been executed, but it has given errors. I'm sorry... I attach the file:

[Resultado.txt](https://github.com/joelsernamoreno/EvilCrow-Keylogger/files/5617711/Resultado.txt)

joelsernamoreno, 2020-11-30:
Write me to my email to fix this

closed 2021-05-11 by joelsernamoreno