joelsernamoreno/EvilCrow-Keylogger #9: Changes are not applied, despite confirmation

scheibleit, 2020-12-13:
The EvilCrow keylogger is a very exciting project. Finally the hardware arrived yesterday and I could try it out. But it seems that my changes are not taken over. For the configuration of the EvilCrow keylogger I use a notebook with a current Kali Linux. I proceeded as described in the Automatic installation section.

**Platformio Core installed**
```
$ python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
$ sudo ln -s ~/.platformio/penv/bin/platformio /usr/local/bin/platformio
$ sudo ln -s ~/.platformio/penv/bin/pio /usr/local/bin/pio
$ sudo ln -s ~/.platformio/penv/bin/piodebuggdb /usr/local/bin/piodebuggdb
```

**Repositories cloned**
```
$ git clone https://github.com/volca/keylogger-pio.git 
$ git clone https://github.com/joelsernamoreno/EvilCrow-Keylogger.git
```

**Files changed**
`$ vim EvilCrow-Keylogger/libraries/Keyboard/src/Keyboard.cpp `
_26: #define kbd_en_us =>  #define kbd_de_de_
`$ vim EvilCrow-Keylogger/blob/master/code/ESP32/ESP32.ino`
_16: const char* ssid = "Keylogger"; => const char* ssid = "KeyloggerX";_

**Jumper set and connected**
_Place the supplied jumper as shown in the photo. And the EvilCrow keylogger connected to the notebook._

**Script started**
`$ sudo ./keylogger-pio/flash.sh`

The script ran and gave a success message ([Output](https://github.com/joelsernamoreno/EvilCrow-Keylogger/files/5684424/output.txt)). After that I stretched out the stick, removed the jumper and plugged it in again with a keyboard connected. The name of the WiFi is still "Keylogger" and the captured characters are special characters. In the web interface I do not see the Live Payload tab.

What do I have to do differently so that the changes are applied?

joelsernamoreno, 2020-12-13:
Hello!

I think the platformio core installation and modifications are correct

How many successes did you receive when you executed ./flash.sh?

You have to receive 3 successes

The script first programs the atmega32u4 to flash ESP32 then, finally it flashes atmega32u4 again with the final code

Maybe you removed the device before the script was finished. Please send me a file with the output of the ./flash.sh command

Thank you!

joelsernamoreno, 2020-12-13:
Hello!

I haven't seen the output of the flash.sh command, I just saw this

Well, you are getting an error when you try to flash ESP32:

Error: Could not find one of 'package.json' manifest files in the package

To solve the error, do is:

1.- Go to the keylogger-pio/esp32 directory
2.- Edit the platformio.ini file and delete these lines:

platform_packages = tool-esptoolpy@https://github.com/AprilBrother/esptool.git#keylogger

3.- Run ./flash.sh

Thanks!

scheibleit, 2020-12-13:
Hello,
thank you for the quick reply. I have removed the two lines and tried again. Now I get twice successful but once an error. The programming of the ESP32 is successful. But he upload of ATMEGA32U4.ino fails.
```
Auto-detected: /dev/ttyACM0                                                                                                                           
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Waiting for the new upload port...
Error: Couldn't find a board on the selected port. Check that you have the correct port selected. If it is correct, try pressing the board's reset button after initiating the upload.                                                                                                                      
*** [upload] Explicit exit, status 1 
```
Complete output: [output2.txt](https://github.com/joelsernamoreno/EvilCrow-Keylogger/files/5684482/output2.txt)

joelsernamoreno, 2020-12-13:
Hello!

It is not detecting the plate correctly 

Disconnect the device, check that the jumper is making contact correctly and try again

I have also seen that you don't have the UDEV rules of platformio, do this:
https://docs.platformio.org/page/faq.html#platformio-udev-rules

If this doesn't work, the final code can be uploaded with the Arduino IDE

If you can't solve the problem, write to me privately on twitter (@JoelSernaMoreno) and I will connect to your computer to fix this 

Thank you!

scheibleit, 2020-12-13:
It worked! With the UDEV rules it worked successfully. Thanks for your quick and straightforward help.
> I have also seen that you don't have the UDEV rules of platformio, do this:
> https://docs.platformio.org/page/faq.html#platformio-udev-rules

closed 2020-12-13 by scheibleit