joelsernamoreno/EvilCrow-Keylogger #10: Errors and warnings during flashing

reynas, 2020-12-22:
Host: Ubuntu 20.04.1
After flashing, it still shows version v1.2 with buttons view log/delete log (livepayload is missing)
Do you have any guidance here? Thanks!

Fulll log:

root@user-HP:~/keylogger-pio# ./flash.sh 
Processing LilyPadUSB (platform: atmelavr; framework: arduino; board: LilyPadUSB)
-----------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing atmelavr
Platform Manager: atmelavr @ 3.0.0 has been installed!
The platform 'atmelavr' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing platformio/toolchain-atmelavr @ ~1.50400.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-atmelavr @ 1.50400.190710 has been installed!
Tool Manager: Installing platformio/framework-arduino-avr @ ~5.1.0
Downloading  [####################################]  100%
Tool Manager: framework-arduino-avr @ 5.1.0 has been installed!
Tool Manager: Installing platformio/tool-avrdude @ ~1.60300.0
Tool Manager: tool-avrdude @ 1.60300.200527 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40001.0
Tool Manager: tool-scons @ 4.40001.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/LilyPadUSB.html
PLATFORM: Atmel AVR (3.0.0) > Arduino LilyPad USB
HARDWARE: ATMEGA32U4 8MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-arduino-avr 5.1.0 
 - tool-avrdude 1.60300.200527 (6.3.0) 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting programmer.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/LilyPadUSB/src/programmer.ino.cpp.o
Archiving .pio/build/LilyPadUSB/libFrameworkArduinoVariant.a
Indexing .pio/build/LilyPadUSB/libFrameworkArduinoVariant.a
Compiling .pio/build/LilyPadUSB/FrameworkArduino/CDC.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/HardwareSerial0.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/HardwareSerial1.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/HardwareSerial2.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/HardwareSerial3.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/PluggableUSB.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/Print.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/USBCore.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/WInterrupts.c.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/WString.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/abi.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/hooks.c.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/main.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/new.cpp.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/wiring.c.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/wiring_analog.c.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/wiring_digital.c.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/wiring_pulse.S.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/LilyPadUSB/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/LilyPadUSB/libFrameworkArduino.a
Indexing .pio/build/LilyPadUSB/libFrameworkArduino.a
[... 235 lines cut ...]

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file ".pio/build/LilyPadUSB/firmware.hex"
avrdude: writing flash (28616 bytes):

Writing | ################################################## | 100% 2.27s

avrdude: 28616 bytes of flash written
avrdude: verifying flash memory against .pio/build/LilyPadUSB/firmware.hex:
avrdude: load data flash data from input file .pio/build/LilyPadUSB/firmware.hex:
avrdude: input file .pio/build/LilyPadUSB/firmware.hex contains 28616 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.33s

avrdude: verifying ...
avrdude: 28616 bytes of flash verified

avrdude: safemode: Fuses OK (E:CE, H:D8, L:FF)

avrdude done.  Thank you.

===================================================================== [SUCCESS] Took 19.73 seconds =====================================================================

reynas, 2020-12-22:
Resolution for https://github.com/joelsernamoreno/EvilCrow-Keylogger/issues/9 helped as well.

closed 2020-12-22 by reynas

joelsernamoreno, 2020-12-22:
Hello!
Have you solved this or do you need help?

reynas, 2020-12-22:
Hi
I solved this with the help of your comments on the issue #9 
Everything works smoothly now, thanks a lot!