joelsernamoreno/EvilCrow-Keylogger #12: Errors during Flash

auyx, 2021-03-03:
Hi I'm getting the following during the flashing process and I cant figure out what the problem is. Hoping you could help. 

auyx@auyx-HP-635-Notebook-PC:~/keylogger-pio$ ./flash.sh
Processing LilyPadUSB (platform: atmelavr; framework: arduino; board: LilyPadUSB)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/LilyPadUSB.html
PLATFORM: Atmel AVR (3.2.0) > Arduino LilyPad USB
HARDWARE: ATMEGA32U4 8MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-arduino-avr 5.1.0 
 - tool-avrdude 1.60300.200527 (6.3.0) 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting programmer.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/LilyPadUSB/src/programmer.ino.cpp.o
Linking .pio/build/LilyPadUSB/firmware.elf
Checking size .pio/build/LilyPadUSB/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.2% (used 542 bytes from 2560 bytes)
Flash: [==        ]  16.0% (used 4580 bytes from 28672 bytes)
Configuring upload protocol...
AVAILABLE: avr109
CURRENT: upload_protocol = avr109
Looking for upload port...
Auto-detected: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Waiting for the new upload port...
Uploading .pio/build/LilyPadUSB/firmware.hex

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file ".pio/build/LilyPadUSB/firmware.hex"
avrdude: writing flash (4580 bytes):

Writing | ################################################## | 100% 0.44s

avrdude: 4580 bytes of flash written
avrdude: verifying flash memory against .pio/build/LilyPadUSB/firmware.hex:
avrdude: load data flash data from input file .pio/build/LilyPadUSB/firmware.hex:
avrdude: input file .pio/build/LilyPadUSB/firmware.hex contains 4580 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.15s

avrdude: verifying ...
avrdude: 4580 bytes of flash verified

avrdude: safemode: Fuses OK (E:CE, H:D8, L:FF)

avrdude done.  Thank you.

[... 95 lines cut ...]
* CLI  > platformio lib search "header:HID.h"
* Web  > https://platformio.org/lib/search?query=header:HID.h
*
*************************************************************

 #include "HID.h"
          ^~~~~~~
compilation terminated.
Compiling .pio/build/LilyPadUSB/libe71/SD/utility/Sd2Card.cpp.o
*** [.pio/build/LilyPadUSB/lib91c/Mouse/Mouse.cpp.o] Error 1
.pio/libdeps/LilyPadUSB/SD/src/utility/Sd2Card.cpp:31:12: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

   #include <SPI.h>
            ^~~~~~~
compilation terminated.
*** [.pio/build/LilyPadUSB/libe71/SD/utility/Sd2Card.cpp.o] Error 1
========================== [FAILED] Took 2.06 seconds ==========================

joelsernamoreno, 2021-03-03:
Hello!

You have keylogger-pio and EvilCrow-Keylogger in the same directory?

Can you run flash.sh without verbose, export the output to a text file and send it to me?

Thanks

closed 2021-03-03 by auyx

joelsernamoreno, 2021-03-03:
Fix?

auyx, 2021-03-03:
I misread the instructions and moved EvilCrow-Keylogger into keylogger-pio. I shouldn't play with this stuff at 2am.