joelsernamoreno/EvilCrow-Keylogger #14: All Three Successes, SSID changed OK, but still can't change keyboard Layout

Audr1un, 2022-01-10:
I have apllied these changes in files:
ESP32.ino:
const char* ssid = "Keylogger"; ==> const char* ssid = "Keyla";

Keyboard.cpp:
#define kbd_en_us ==> #define kbd_pt_pt
(also, tested too with es_es and cz_cz. No effect)

No errors when run the flash.bat, no relevant (apparently) warnings, as we can see in log:
C:\Users\philipe.martins\KL\keylogger-pio>C:\Users\philipe.martins\.platformio\penv\Scripts\platformio.exe run -t upload -d programmer
Processing LilyPadUSB (platform: atmelavr; framework: arduino; board: LilyPadUSB)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/LilyPadUSB.html
PLATFORM: Atmel AVR (3.4.0) > Arduino LilyPad USB
HARDWARE: ATMEGA32U4 8MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - tool-avrdude 1.60300.200527 (6.3.0)
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
Converting programmer.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\LilyPadUSB\src\programmer.ino.cpp.o
Linking .pio\build\LilyPadUSB\firmware.elf
Checking size .pio\build\LilyPadUSB\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.2% (used 542 bytes from 2560 bytes)
Flash: [==        ]  16.0% (used 4580 bytes from 28672 bytes)
Configuring upload protocol...
AVAILABLE: avr109
CURRENT: upload_protocol = avr109
Looking for upload port...
Auto-detected: COM4
Forcing reset using 1200bps open/close on port COM4
Waiting for the new upload port...
Uploading .pio\build\LilyPadUSB\firmware.hex

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file ".pio\build\LilyPadUSB\firmware.hex"
avrdude: writing flash (4580 bytes):

Writing | ################################################## | 100% 0.38s

avrdude: 4580 bytes of flash written
avrdude: verifying flash memory against .pio\build\LilyPadUSB\firmware.hex:
avrdude: load data flash data from input file .pio\build\LilyPadUSB\firmware.hex:
avrdude: input file .pio\build\LilyPadUSB\firmware.hex contains 4580 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.06s

[... 342 lines cut ...]
avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file ".pio\build\LilyPadUSB\firmware.hex"
avrdude: writing flash (28210 bytes):

Writing | ################################################## | 100% 2.28s

avrdude: 28210 bytes of flash written
avrdude: verifying flash memory against .pio\build\LilyPadUSB\firmware.hex:
avrdude: load data flash data from input file .pio\build\LilyPadUSB\firmware.hex:
avrdude: input file .pio\build\LilyPadUSB\firmware.hex contains 28210 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.37s

avrdude: verifying ...
avrdude: 28210 bytes of flash verified

avrdude: safemode: Fuses OK (E:CE, H:D8, L:FF)

avrdude done.  Thank you.

======================================================================================================= [SUCCESS] Took 31.84 seconds =========================================================================================

Please Help, I am burning my sanity for almost three days.
In time, before I get trapped in this, I was wodering if I could change the scan codes of keys / and ? in english keyboard, cause the keylogger works like charm in a PT_BR keyboard, except the InternationalKey1 (HID 0x87), which have the same scan code of the modifier key KEY_RIGHT_GUI. Anyway, in a short answer, is easy and simple just make this change or I would need to make fundamental (and brutal) changes in the code?

joelsernamoreno, 2022-01-10:
Hi!

Last time I changed the layout on my device it was working correctly, I will check this this week. To change the layout you just need to change Keyboard.cpp as indicated in the documentation 

About changing the scan codes... this is simple. In the sketch you can find the modifiers, change this with the values you need. The keymaps can be found in EvilCrow-Keylogger/libraries/Keyboard/src/*

For example, for the layout en_us, open en_us.h and change the values you need

Audr1un, 2022-01-13:
Hello again!

I have changed the Keyboard.cpp to pt_pt and delete all the others languages libraries, including en_us. This stubborn device refuse to abandon the en_us language. The log shows NO errors so far.

If you are too much busy, just give me some of your thoughts what could cause this and I will happily search how to solve it.

joelsernamoreno, 2022-01-13:
Hi!

Can you check that the PT layout is correct? I have never been able to check that it is correct, maybe this is the problem

Here is a list of layouts from another project, maybe you can find the correct one here:

https://github.com/insecurityofthings/uC_mousejack/blob/master/tools/keymap.py

You can modify pt_pt.h and change the layout

Another option is to upload the atmega32u4.ino sketch with the Arduino IDE, but I haven't seen any errors in your platformio log. Possibly the problem is that the PT layout is incorrect

Audr1un, 2022-01-17:
Hello!
Well,  I messed up a little with the en_us layout just to see (and prove my point of he is not only evil, but an insurgent crow too), and yet no effect, the keyboard's output and the log didn't show the changes: 'n' is still 'n', 'o' still 'o', 'p' still 'p' and the keyscan 0x87 still don't show anything.
[image]

When I tried to upload atmega32u4.ino with Arduino IDE, some errors was given:

Arduino: 1.8.19 (Windows 10), Board: "LilyPad Arduino USB"

C:\Users\Audr1un\KL\EvilCrow-Keylogger\code\ATMEGA32U4\ATMEGA32U4.ino: In member function 'virtual void KbdRptParser::OnKeyUp(uint8_t, uint8_t)':

ATMEGA32U4:68:12: error: 'class Keyboard_' has no member named 'rawrelease'; did you mean 'release'?

   Keyboard.rawrelease(key, 0);

            ^~~~~~~~~~
            release

C:\Users\Audr1un\KL\EvilCrow-Keylogger\code\ATMEGA32U4\ATMEGA32U4.ino: In member function 'virtual void KbdRptParser::OnKeyDown(uint8_t, uint8_t)':

ATMEGA32U4:84:12: error: 'class Keyboard_' has no member named 'rawpress'; did you mean 'press'?

   Keyboard.rawpress(key, modifiers);

            ^~~~~~~~
            press

Multiple libraries were found for "hidboot.h"

 Used: C:\Users\Audr1un\Documents\Arduino\libraries\USB_Host_Shield_2.0

 Not used: C:\Program Files (x86)\Arduino\libraries\USB_Host_Shield_2.0

Multiple libraries were found for "Keyboard.h"

 Used: C:\Users\Audr1un\Documents\Arduino\libraries\Keyboard

 Not used: C:\Program Files (x86)\Arduino\libraries\Keyboard

exit status 1

'class Keyboard_' has no member named 'rawrelease'; did you mean 'release'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Audr1un, 2022-01-17:
Ok, I noticed some things, I will edit this comment in no time

joelsernamoreno, 2022-01-17:
Hi!

The rawrelease and rawpress functions were created to make the keylogger work properly

In order to compile atmega32u4.ino with arduino ide you have to copy the Keyboard and USB Host libraries (this is in the libraries folder) in the Arduino library directory

Audr1un, 2022-01-17:
Finally I did understand.
In the LOG the mess take effect, the 0x87 make the promissed '/' character appears. But I'm still struggling because when I press that key, the monitor don't show anything, the EvilCrow log and then discards. In other words, in the log, I can make the pt_br layout to work fine, but the feedback in the screen works like a en_us layout.

Your library was expect to work like that? How can I make the output of the EvilCrow to be the same as the input?

joelsernamoreno, 2022-01-17:
Read atmega32u4.ino. Here you can find Onkeyup and Onkeydown. If you look at the code, the key you press and release on your keyboard is looked up in asciimap. This is then saved to the SD and displayed on the web. The key you press on the keyboard has to be the same as the one shown on the web

I don't know the PT keyboard, to do this do you do a key combination or just press a key?

[image]

Audr1un, 2022-01-18:
Web do you mean when I access the wifi from the ESP32? In the web it's shown OK, the problem is just the feedback from the user's input in their screen monitor.
The key is the same as english layout, / is raw, ? is when I press the Shiftkey.
[image]

I read the atmega32u4.ino and digged some more, are you sure that's the problem is not for the fact of the scancode is 0x87?
the keyboard file, in the line https://github.com/joelsernamoreno/EvilCrow-Keylogger/blob/master/libraries/Keyboard/src/Keyboard.cpp#L557
admits 0x87 like a modifier

joelsernamoreno, 2022-01-18:
Screen monitor? Do you mean that typing / with your keyboard does not show up on the computer or in the keylogger log?

Audr1un, 2022-01-18:
On the computer of the target. I have discovered yesterday that log works properly with some modifications in the keyboard layout. Now it's just the EvilCrow it's not sending the / key to the computer after log it

joelsernamoreno, 2022-01-18:
Well, then the problem is in sending the key to the computer, not in the log.

Ok, I understand that for / you use SHIFT + /

Right?

This works with rawpress (onkeydown), not rawrelease

Shift is MODIFIERKEY_LEFT_SHIFT (0x02), this should be correct for your keyboard. So... try to press SHIFT + / very slowly, maybe this is the problem

joelsernamoreno, 2022-01-18:
Anyway, in the keyboard picture you sent me it should not be necessary to press Shift to print /

Audr1un, 2022-01-18:
No, look at the keyboard image, the key '/' should apprear without any modifier and the '?' is SHIFT + /

Pressing slowly didn't work

joelsernamoreno, 2022-01-18:
Ok, so how do you have configured / in the layout?

Audr1un, 2022-01-18:
[image]

Audr1un, 2022-01-18:
It's a special key defined in the HID as international key, only present in the japanese and brazilian keyboard. I'm sure not only for that, but because the log works after I changed the layout from 0x38 to 0x87

joelsernamoreno, 2022-01-18:
So I don't really know what the problem is. This should work for /

Audr1un, 2022-01-18:
:(

joelsernamoreno, 2022-01-18:
If it is onkeydown you comment the line Keyboard.rawpress(key, modifiers); and you put Keyboard.print("/") what happens?

Audr1un, 2022-01-18:
Well, 3 different things:
1 - in the computer screen, every key pressed turns 'D', the '/' key and SHIFT + '/' key shows the D too  
2 - in the web log, works properly, except it shows / as   /?Dq
3 - in the sd log, it shows everything normal, except for the SHIFT + /, it shows just '/' (yes, I didnt notice before, the log don't recognize SHIFT + / as ?)

Audr1un, 2022-01-18:
The 'D' is probabily an error, because when I change to Keyboard.print("b"), for example, it's shows 'b' on the screen

closed 2022-01-18 by Audr1un