joelsernamoreno/EvilCrow-Keylogger #19: Log issues

XtramCZ, 2023-08-10:
Some special characters aren't even logging (ě š č ř ž ý á í í é # & etc)

weird encocing issues
this is in the log: some spe&cial &char&cters an}d n}umpad aren}t e@ven w|orkin}]g

im using the cz_cz layout btw

joelsernamoreno, 2023-08-13:
Hello!

I don't know the Russian keyboard, so there will be some errors here

You can modify the wrong keys in the layout. I don't have any Russian keyboard and I can't check this

XtramCZ, 2023-08-13:
Okay, but still there's an issue with the windows key

joelsernamoreno, 2023-08-13:
What do you mean by pressing the key?

If you press the windows key and then release the windows key... this should work.

If you want to press windows + another key, you have to add some lines of code in the firmware 

I'm away from home this month, I can check this later

Anyway, in the atmega32u4 firmware, look for onkeydown. Here you can add an "if" to press GUI + another key

XtramCZ, 2023-08-13:
Yes, I meant press windows + another key
thanks