joelsernamoreno/EvilCrow-Keylogger #20: Not logging Caps or special characters - shift key

squirreltech2, 2023-11-05:
Everything is working great but it's not capturing caps or special characters...it's just logging as if the shift key wasn't pressed...am I missing some setting I need to specify in the code to enable that?

squirreltech2, 2023-11-05:
Tried with two different keyboards and 3 different operating systems on 2 computers, all the same results

squirreltech2, 2023-11-05:
Full log of programming:

┌──(squirrel㉿kaliLaptop)-[~/keylogger-pio]
└─$ ./flash.sh 
Processing LilyPadUSB (platform: atmelavr; framework: arduino; board: LilyPadUSB)
-------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/LilyPadUSB.html
PLATFORM: Atmel AVR (4.2.0) > Arduino LilyPad USB
HARDWARE: ATMEGA32U4 8MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES: 
 - framework-arduino-avr @ 5.1.0 
 - tool-avrdude @ 1.60300.200527 (6.3.0) 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
Converting programmer.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/LilyPadUSB/src/programmer.ino.cpp.o
Linking .pio/build/LilyPadUSB/firmware.elf
Checking size .pio/build/LilyPadUSB/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.2% (used 542 bytes from 2560 bytes)
Flash: [==        ]  16.0% (used 4580 bytes from 28672 bytes)
Configuring upload protocol...
AVAILABLE: avr109
CURRENT: upload_protocol = avr109
Looking for upload port...
Auto-detected: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Waiting for the new upload port...
Uploading .pio/build/LilyPadUSB/firmware.hex

Connecting to programmer: .                                                                                        
Found programmer: Id = "CATERIN"; type = S                                                                         
    Software Version = 1.0; No Hardware Version given.                                                             
Programmer supports auto addr increment.                                                                           
Programmer supports buffered memory access with buffersize=128 bytes.                                              
                                                                                                                   
Programmer supports the following devices:                                                                         
    Device code: 0x44                                                                                              
                                                                                                                   
avrdude: AVR device initialized and ready to accept instructions                                                   
                                                                                                                   
Reading | ################################################## | 100% 0.00s                                          

avrdude: Device signature = 0x1e9587 (probably m32u4)                                                              
avrdude: reading input file ".pio/build/LilyPadUSB/firmware.hex"                                                   
avrdude: writing flash (4580 bytes):                                                                               
                                                                                                                   
Writing | ################################################## | 100% 0.37s                                          

avrdude: 4580 bytes of flash written                                                                               
avrdude: verifying flash memory against .pio/build/LilyPadUSB/firmware.hex:                                        
avrdude: load data flash data from input file .pio/build/LilyPadUSB/firmware.hex:                                  
avrdude: input file .pio/build/LilyPadUSB/firmware.hex contains 4580 bytes                                         
avrdude: reading on-chip flash data:                                                                               
                                                                                                                   
Reading | ################################################## | 100% 0.06s                                          

avrdude: verifying ...                                                                                             
avrdude: 4580 bytes of flash verified                                                                              
                                                                                                                   
avrdude: safemode: Fuses OK (E:CE, H:D8, L:FF)                                                                     
                                                                                                                   
avrdude done.  Thank you.                                                                                          
[... 170 lines cut ...]
                                                                                                                   
Reading | ################################################## | 100% 0.00s                                          

avrdude: Device signature = 0x1e9587 (probably m32u4)                                                              
[... cut ...]
avrdude: safemode: Fuses OK (E:CE, H:D8, L:FF)                                                                     
                                                                                                                   
avrdude done.  Thank you.                                                                                          
                                                                                                                   
=========================================== [SUCCESS] Took 5.60 seconds ===========================================

joelsernamoreno, 2023-11-07:
Hello!

Are you pressing the key combination too fast?

Try pressing and holding the shift key, then press and release a key (e.g. the letter a), then release the shift key

Let me know if this works and I will give you a solution

squirreltech2, 2023-11-08:
did not work... this is what I typed on the keyboard connected to the keylogger: "This is a test of CAPS and Specal Characters 1234567890 !@#$%^&*()" and this is what the log shows on the web page and on the actual log file in the memory card: "this is a test of caps and special characters 1234567890 1234567890"

So it's logging but not exactly useful. For info I'm in North America and we don't have Alt GR keys on keyboards. I've tried 3 keyboards and all the same result on two different computers so I'm 95% sure it's something in the code or keyboard files.

joelsernamoreno, 2023-11-15:
This is very strange. What is the layout of your keyboard and what is the layout you are using in the keylogger? This is the first time someone reports this problem

LiquidGas-air, 2024-03-14:
the same issue, capslock key is not registered (tried on different keyboards with EN_US layout)