joelsernamoreno/EvilCrow-Keylogger #22: Error with connection to keylogger esp32 module

lowkickson, 2023-11-24:
Hi, i just bough your keylogger and i have problem with flashing, I think i have all jumpers connected well.I have installed Arduino IDE with esp32 board (2.0.11) and i have added additional boards manager (https://dl.espressif.com/dl/package_esp32_index.json) and still when i want to upload ESP32.ino i have following error

[image]

and this is my jumper connection.

[image]

Can you please help me with this or if is possible to send me pinout connection between keylogger and FTDI.

XtramCZ, 2023-12-22:
I have the same issue

joelsernamoreno, 2023-12-22:
Hello @lowkickson @XtramCZ 

Why don't you try "Automatic installation"? This is much faster and easier

XtramCZ, 2023-12-22:
I was doing automatic instalation. I fixed it tho, I just did it on linux instead of windows and it works

joelsernamoreno, 2023-12-22:
@XtramCZ For this to work on windows you will have to install the usb-uart drivers :)

XtramCZ, 2023-12-23:
even on automatic instalation?